arduino-libraries/LiquidCrystal #1: make startup delay optional

tablatronix, 2016-03-31:
` delayMicroseconds(50000); ` is blocking on esp8266, and probably not needed for most instances anyway.

altishchenko, 2016-05-21:
That is what LCD datasheet tells, unfortunately. They are not very fast devices to startup, though some of them do allow you to shorten the delay to 20-30ms as opposed to standard 40. Still this delay is mandatory after you switched on the power on the display, if you remove it - garbage happens sporadically and lcd ends up in misconfigured mode. What ESP type are you using? Can you show bit of your code?

tablatronix, 2016-05-21:
I do not, pretty sure I was just using the examples, regardless of what the datasheet says, `delaymicroseconds` is blocking and causes issues with the native wifi autoconnect.

Using esp 12e, nodemcu, adafruit huzzah,

I just unwired my lcd, using i2c lib now.

altishchenko, 2016-05-22:
Sounds like catch22 to me.
Just for a try I removed those delays from init()/begin() sequence in my
fork of the library and things became very unstable - 50/50 and even worse.
I'll try to couple the LCD with my NodeMCU and see what issues you might
have there. You are not using ATA command set for Node, you burning it
directly, right?

i2c is a good solution, my LCD is a standard one and does not support i2c
and as such takes a lot of pins, so i am trying to make it work over
74HC595 shift register - at least that will cover the databus.

С уважением,
Александр Тищенко
+7 (916) 704-6157
atishch@gmail.com

tablatronix, 2016-05-22:
At least change to delay(50)( delay is not blocking ), but even then , i can handle startup delays in my own code, if i already have 1000ms delays for wifi I do not need another 50ms kicking around in a lib beyond my control.

I remove the line entirely for my use, hence the add an option. 50ms is a long time when it might not be necessary for all modules, and it is beyond your control when it is inside a library, hence the request for some option to disable or global to modify.

altishchenko, 2016-05-23:
Does changing to delay(50) works for you?
The major problem with the LCD is the strict timing of its init sequence... 
Did you try postponing the lcd creation to some later point in the code? Like:
`
LiquidCrystal lcd;
int initialized = 0;
...
void loop() {
   if (!initialized) {
      lcd(12, 11, 5, 4, 3, 2);
      initialized = 1;
   }
   ...
}
`

altishchenko, 2016-05-24:
The reason I am asking - my LCD refuses to work if I change delayMicroseconds(50000) to delay(50)...

tablatronix, 2016-05-24:
interesting, ill try to test when i get a chance, i just removed it entirely, so I am not sure.

altishchenko, 2016-05-25:
Surprisingly enough, it does work with 'lazy' setup. Say, if you postpone the LCD initialization till at least setup(). Like that:

`LiquidCrystal *lcd;`
`void setup() {`
`   lcd = new LiquidCrystal(13,12,11,10,9,8);`
`   lcd->begin(16,2);`
`   ...`
`}`

I even went ahead and tried it with yield() based wait function like this:
`void usdelay(unsigned long us) {`
`  unsigned long stop = micros() + us;`
`  while (micros() < stop) {`
`     yield();`
`  }`
`}`
This one also **doesn't** work in pre-setup stage, but does in 'lazy' mode.
And the important point here - the display initializes consistently.