arduino-libraries/LiquidCrystal #41: LiquidCrystal causing ESP32-S2-Wrover-I to crash.

ammartahircheema, 2020-12-21:
ESP32-s2 is crashing on LCD initialization. Code is 'Hello World' from examples.

Debug Logs is as follow.
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x10 (RTCWDT_RTC_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x8
load:0x3ffe6108,len:0x620
load:0x4004c000,len:0xa38
load:0x40050000,len:0x2848
entry 0x4004c190

`#include <LiquidCrystal.h>

// initialize the library by associating any needed LCD interface pin
// with the arduino pin number it is connected to
const int rs = 7, en = 6, d4 = 5, d5 = 4, d6 = 3, d7 = 8;
LiquidCrystal lcd(rs, en, d4, d5, d6, d7);

void setup() {
  // set up the LCD's number of columns and rows:
  lcd.begin(16, 2);
  // Print a message to the LCD.
  lcd.print("hello, world!");
}

void loop() {
  // set the cursor to column 0, line 1
  // (note: line 1 is the second row, since counting begins with 0):
  lcd.setCursor(0, 1);
  // print the number of seconds since reset:
  lcd.print(millis() / 1000);
}
`

ammartahircheema, 2020-12-21:
Update. Crash only occurs when I initialize LCD on global scope.
"LiquidCrystal lcd(7, 6, 5, 4, 3, 8);" keeping this statement inside a function somehow resolves this issue. but i need lcd variable accessible in both setup and loop function...

proboscide99, 2021-08-05:
I'm having the same issue on a WROOM after updating the esp32 core to current version.
Looking at the library file, I see that the constructor calls 'begin(16,1)'.

Calling 'begin(16,1)' at the end of the constructor leads to an endless delay because -it seems that- the 'delayMicroseconds()' function does not work yet at that point of the initialization. For some reason, it used to work with an older core (V3.3).

The program gets halted even before being able to output anything on the Serial.

Commenting out the 'begin(16,1)' call from the 'LiquidCrystal::init' function in 'LiquidCrystal.cpp' and placing it in the sketch (where it should be already, with proper display size) solves the issue, because the 'delayMicroseconds()' function will work.

Alessandro, Italy

RudyFiero, 2021-11-09:
I asked for help on StackExchange-Arduino. Commenting out the unnecessary  begin(16, 1);   resolved the problem.

[https://arduino.stackexchange.com/a/86565/47766]

the constructor calls LiquidCrystal::init().
LiquidCrystal::init() calls LiquidCrystal::begin().
LiquidCrystal::begin() uses delays.

Those delays may depend on things that aren't configured yet when the constructors are running, and lock up causing a WDT.

**Remove the call to begin() in the LiquidCrystal::init() function** and it should at least get past running the constructor so it gets to run the rest of your code.

_Quite why they have that call in there is beyond me. Rule number one of writing an Arduino library: never perform any operations more strenuous than assigning variables in the constructor as resources you may rely on may not have been initialized yet._

```
void LiquidCrystal::init(uint8_t fourbitmode, uint8_t rs, uint8_t rw, uint8_t enable,
			 uint8_t d0, uint8_t d1, uint8_t d2, uint8_t d3,
			 uint8_t d4, uint8_t d5, uint8_t d6, uint8_t d7)
{
  _rs_pin = rs;
  _rw_pin = rw;
  _enable_pin = enable;
  
  _data_pins[0] = d0;
  _data_pins[1] = d1;
  _data_pins[2] = d2;
  _data_pins[3] = d3; 
  _data_pins[4] = d4;
  _data_pins[5] = d5;
  _data_pins[6] = d6;
  _data_pins[7] = d7; 

  if (fourbitmode)
    _displayfunction = LCD_4BITMODE | LCD_1LINE | LCD_5x8DOTS;
  else 
    _displayfunction = LCD_8BITMODE | LCD_1LINE | LCD_5x8DOTS;
  
  //begin(16, 1);                   **<<<<<<<<<<<<<<<< Remove this <<<<<<<<<<<<<<<<**
}

```

per1234, 2021-11-09:
The problem with the constructor was previously reported at https://github.com/arduino-libraries/LiquidCrystal/issues/11

https://github.com/arduino-libraries/LiquidCrystal/issues/11 is odd in that it was originally a pull request https://github.com/arduino/Arduino/pull/87, but was somehow converted to an issue during the transfer of the library from its original home bundled with the Arduino IDE repository to this dedicated repository. The proposed changes are in this branch:
https://github.com/arduino-libraries/LiquidCrystal/tree/pr_87
with the constructor fix in this commit:
https://github.com/arduino/Arduino/pull/87/commits/f82ec65a0d6e22f39266e91b85cb615df0780b3f