arduino-libraries/LiquidCrystal #49: LiquidCrystal_I2C lcd(0x27, 16, 2); Causes problem on NodeMCU

paulwrath1223, 2021-05-18:
I'm new to github and arduino, but using _LiquidCrystal_I2C lcd(0x27, 16, 2);_ causes an error on NodeMCU boards (and maybe others) but not on Mega 2560. The workaround is to replace it with _LiquidCrystal_I2C lcd(0x27);_ and add _lcd.begin(16,2);_ in setup. Maybe this is obvious, but it confused me a lot, and the error message was very unhelpful. ("Invalid Converson from ‘int’ to ‘t_backlighPol" [-fpermissive]")

per1234, 2021-05-18:
Hi @paulwrath1223. Thanks for your interest in this open source project. I can see from your code that you're using a library for I2C displays. This library does not provide support of any kind for I2C displays and does not have an API resembling that which you have shared.

Since it's clear that this report is related to some other library, and thus off topic here, I'll go ahead and close this.

closed 2021-05-18 by per1234

paulwrath1223, 2021-05-18:
Yeah sorry, that was meant for the LiquidCrystal_I2C Library.