arduino-libraries/LiquidCrystal #65: Broken link in examples/HelloWorld.ino

drf5n, 2023-02-07:
The comments at 

https://github.com/arduino-libraries/LiquidCrystal/blob/master/examples/HelloWorld/HelloWorld.ino#L39

have a broken link to https://www.arduino.cc/en/Tutorial/LibraryExamples/HelloWorld

See discussion at https://forum.arduino.cc/t/lcd-example-in-arduino/1086684  which suggests using the tutorial at https://docs.arduino.cc/learn/electronics/lcd-displays#hello-world-example instead.

per1234, 2023-02-07:
Hi @drf5n. Thanks for your report.

>The comments at
>
>https://github.com/arduino-libraries/LiquidCrystal/blob/master/examples/HelloWorld/HelloWorld.ino#L39
>
>have a broken link

The link works fine for me. Please confirm whether or not clicking this link takes you to the LiquidCrystal tutorial page:

https://www.arduino.cc/en/Tutorial/LibraryExamples/HelloWorld

>the tutorial at https://docs.arduino.cc/learn/electronics/lcd-displays#hello-world-example instead. (The code there also has the same broken link)

I have submitted a fix for this: https://github.com/arduino/docs-content/pull/827

drf5n, 2023-02-07:
It looks like there is now a redirect from 

https://www.arduino.cc/en/Tutorial/LibraryExamples/HelloWorld

to 

https://docs.arduino.cc/learn/electronics/lcd-displays 

```
% curl -i https://www.arduino.cc/en/Tutorial/LibraryExamples/HelloWorld
HTTP/2 302 
date: Tue, 07 Feb 2023 14:06:37 GMT
location: https://docs.arduino.cc/learn/electronics/lcd-displays
cache-control: private, max-age=0, no-store, no-cache, must-revalidate, post-check=0, pre-check=0
expires: Thu, 01 Jan 1970 00:00:01 GMT
set-cookie: __cf_bm=GPkXk13G3RJ6CAdd6_Rk5YMVx6qe.mrdjkSx3.GMoDI-1675778797-0-AeQBFMwd83ymd8SAqWwwAoQ52wl7nxIYDTgapy1SBQtk26K7o552yHYUKoCUSWI2DxZ73jmpE09S01EDGhvdwMc=; path=/; expires=Tue, 07-Feb-23 14:36:37 GMT; domain=.arduino.cc; HttpOnly; Secure; SameSite=None
server: cloudflare
cf-ray: 795cafec388081e5-IAD
alt-svc: h3=":443"; ma=86400, h3-29=":443"; ma=86400

So the link is repaired, not by editing the code in HelloWorld.ino, but by the redirect on the server end.

per1234, 2023-02-07:
Thanks for confirming. Even though the URL is working, best practices are to avoid relying on redirects since they are not guaranteed to work forever.

So that you can receive the proper attribution for spotting this problem, would you care to submit a pull request to update this URL in the examples (and the others as well if you want to go the extra mile) @drf5n?

drf5n, 2023-02-07:
I used the github online edit/fork tool to make pull request https://github.com/arduino-libraries/LiquidCrystal/pull/66

closed 2023-02-08 by per1234